thegecko/webbluetooth #2: Update device advert scanning functionality

thegecko, 2018-01-03:
The specification outlines a (not yet stable) ability to watch a device's adverts:

https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothdevice-watchadvertisements

This likely needs quite a refactor of the noble adapter code to continually scan for devices and read the adData for a specific device.

Implement:

 - BluetoothDevice.watchAdvertisements()
 - BluetoothDevice.unwatchAdvertisements()
 - BluetoothDevice.watchingAdvertisements

Remove:

 - BluetoothDevice.adData

Update:

 - Eddystone example

m4-miranda, 2019-03-12:
Hi, fairly new to the project and the spec, is this functionality built into Chrome? How would one go about contributing?

thegecko, 2019-03-12:
The [implementation status](https://github.com/WebBluetoothCG/web-bluetooth/blob/master/implementation-status.md#scanning-api) suggests this isn't supported in Chrome yet.
That doesn't mean you can't implement it following the specification, but there's no way to then compare it with another implementation.

tavrez, 2024-12-01:
I need to read a payload of a services on a non-connectable device, is this the method needed for my work?