thegecko/webbluetooth #5: getPrimaryService and getCharacteristic fail with numeric short uuid

vshymanskyy, 2018-03-26:
It fails to find the service/characteristic, but when I change `0x2a2b` param to `UUID_SHORT('2a2b')`, it works.
Also tested the same code with WebBluetooth in Chrome and it works as expected with `0x2a2b`.

thegecko, 2018-03-27:
Hmm, all of the methods should be converting the uuid to it's canonical form using this function:

https://github.com/thegecko/webbluetooth/blob/master/src/helpers.ts#L257

I've just tested the following with it and all looks fine:

```javascript
getCanonicalUUID(0x2a2b); // "00002a2b-0000-1000-8000-00805f9b34fb"
getCanonicalUUID("2a2b"); // "00002a2b-0000-1000-8000-00805f9b34fb"
```

It's possible it's not being called everywhere, I'll investigate further.

thegecko, 2018-03-27:
@vshymanskyy I can't seem to recreate your issue. I have a device here which advertises the short UUID for a service and it is correctly found using the following:

```javascript
bluetooth.requestDevice({
    filters:[{ services:[ 0xfe8e ] }]
})
```

Internally both the advertised UUID and search UUID are being converted to the canonical UUID. Could you outline exactly how you are searching?

vshymanskyy, 2018-03-27:
@thegecko ok, thanks for your quick response. It seems to happen only for `optionalServices` I have.
BTW here is my library for MiBand 2 (with Node.js and Browser examples):
https://github.com/vshymanskyy/miband-js

And here is my temporary workaround: https://github.com/vshymanskyy/miband-js/blob/master/src/miband.js#L11

vshymanskyy, 2018-03-27:
@thegecko BTW, you call `getCharacteristicUUID` for every item here: https://github.com/thegecko/webbluetooth/blob/master/src/service.ts#L126
Probably can be optimized a little bit by converting it once and then filtering...

thegecko, 2018-03-28:
@vshymanskyy I think I found an issue, would you mind testing with PR #6 ?

vshymanskyy, 2018-03-28:
One more observation.
`const UUID_SERVICE_HEART_RATE = '180d'` works
while
`const UUID_SERVICE_HEART_RATE = 0x180d` doesn't.

OK, let me check your PR

thegecko, 2018-03-28:
They should both work :/. What OS and node version are you running on?

vshymanskyy, 2018-03-28:
This is Linux, Node v8.9.4

thegecko, 2018-03-28:
Right, time to fire up some virtual machines...

vshymanskyy, 2018-03-28:
FYI, the Chrome implementation fails to find short string representation:
Failed to execute 'requestDevice' on 'Bluetooth': Invalid Service name: '1800'.
-> It must be a valid UUID alias (e.g. 0x1234), UUID (lowercase hex characters e.g. '00001234-0000-1000-8000-00805f9b34fb'), or recognized standard name from https://www.bluetooth.com/specifications/gatt/services

So the only usable workaround for me is the `UUID_SHORT` wrapper ;)

thegecko, 2018-04-03:
@vshymanskyy  I've merged a fix which should make the mi band more discoverable on Linux. It seems noble won't match canonical UUIDs and short UUIDs during it's search on Linux and when the package was checking the UUIDs instead of noble, it would only see the "first" advert from the device which didn't always have the service UUIDs in it. This PR should make that all more reliable and I tested it using your mi band script (I happen to own one!) on a Linux VM as well as a MacOS host.

closed 2018-04-03 by thegecko

vshymanskyy, 2018-04-03:
@thegecko thanks for your efforts. But it still doesn't work for me. And I believe I have tracked down the bug now :) I'll post an update in some 15-20 minutes.

vshymanskyy, 2018-04-03:
The main problem appears, when I try to  call `miband1.getCharacteristic(0x2a2b)`

'getCharacteristic', 'getPrimaryService' should allow:

- Valid UUID alias (e.g. 0x1234)
- Full UUID (lowercase hex characters e.g. '00001234-0000-1000-8000-00805f9b34fb')
- Recognized standard name from https://www.bluetooth.com/specifications/gatt/services

It appears that when I supply `0x2a2b`, it's mapped back to `"current_time"` here:
https://github.com/thegecko/webbluetooth/blob/master/src/helpers.ts#L281
and then it fails to convert to canonical form completely.

I have removed my workaround in the master branch of MiBand JS, so you can verify it.
The Chrome still works fine after the change.

thegecko, 2018-04-03:
Thanks, got it now. Enums in TypeScript are compiled to allow a reverse lookup so not only does it allow you to find the number from a string, but also the string from a number. I'm trying two fixes for this...

thegecko, 2018-04-03:
Have created two fixes:

https://github.com/thegecko/webbluetooth/tree/d/enums
https://github.com/thegecko/webbluetooth/tree/d/enums2

I think I'll use the last one, however as it changes the output code the least:

https://github.com/thegecko/webbluetooth/pull/7

vshymanskyy, 2018-04-03:
@thegecko both look good. The second might be better ;) it depends on whether the reverse lookup is needed in some cases.

thegecko, 2018-04-03:
As I expose it I figured I'd keep it the same type in case anyone is using the reverse lookup :)

thegecko, 2018-04-03:
[1.0.6 published](https://www.npmjs.com/package/webbluetooth)

vshymanskyy, 2018-04-03:
@thegecko updated. It works now. Thanks!

thegecko, 2018-04-03:
No problem, thanks for helping make this package better

viktornord, 2018-07-06:
@thegecko Hello and thank you for your lib 
I Created a [repo](https://github.com/viktornord/ble) based on the lib for communicating with Mi Band by @vshymanskyy which works perfect on MacOS but when I tried on Linux (Ubuntu Mate) I can connect, get base info, send notifications but heart rate measurement is not working when I send 3 write commands to heart rate data characteristic. I am using noble directly (without webbluetooth lib).
I investigated your code and noticed that you are also using noble but still can't get why it's not working when using noble without this webbluetooth adapter.
Webbluetooth works well but that's another layer of abstraction. Any ideas why using noble directly is not working?

thegecko, 2018-07-06:
Unfortunately not, you could ask about it on the noble repo?

PR #6 (2018-04-03): Switch to noble service scanning
Noble service scanning seems to now support canonical UUIDs and short UUIDs and vice-versa (on MacOS at least)

This PR switches to use the noble scanning methods which finds lesser-advertised services.
Also made it more aggresive.

Also includes some minor example updates.

@vshymanskyy this should fix #5