thegecko/webbluetooth #8: Allow selection of a discovered device

viktornord, 2018-07-03:
Currently filter should be specified to select some bluetooth peripheral device.
Is there a way to select a certain device out of discovered ones when using this lib in node.js? It is important when I don't know beforehand which device should be connected and user interaction is needed.
For example I want to connect Mi Band 2 and there are several Mi Band 2 devices nearby but I don't know exactly which one should be connected.

thegecko, 2018-07-03:
Please refer to the `selector` demo, specifically how the a function is injected to allow you to handle the device choosing:

https://github.com/thegecko/webbluetooth/blob/master/examples/selector.js#L63

viktornord, 2018-07-03:
wow, sorry, didn't notice. Thanks!

closed 2018-07-03 by thegecko