thegecko/webbluetooth #9: Memory leak after reconnect

uri71, 2018-07-17:
Hi! 
I create the application using your library and met one issue. 
When I try to reconnect to the gatt several times: 
device.gatt.connect()
I got memory leak warning 
(node:5619) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 disconnect listeners added. Use emitter.setMaxListeners() to increase limit

I do it in the loop: try to connect to the device for 2 minutes, then 3 sec delay and try it again - until the connection is established

As I understood it happens because listeners on ''connect"  and 'disconnect' added each time in NobleAdapter.connect() method. 
Could you please prompt: is there the possibility to clear these listeners?
And when we connect to the gatt: how long should be this connection? E.g. : if I want to connect while the connection is established - how long should be the connection time?
Or maybe my connection logic is wrong?
Thank in advance

thegecko, 2018-07-21:
Thanks, I'll have a look into disconnecting these event listeners.

thegecko, 2018-07-21:
@uri71 Can you tell me if this commit fixes your issue: https://github.com/thegecko/webbluetooth/commit/5da4236a0cb84bca6bd0a766c44ae6182ea6c0f0

viktornord, 2018-07-24:
@thegecko likely this won't fix the problem upon unexpected disconnect, when you receive an event `disconnect`, because listeners are removed only when disconnect method. I think they should also be cleaned up here https://github.com/thegecko/webbluetooth/commit/5da4236a0cb84bca6bd0a766c44ae6182ea6c0f0#diff-080186fdfa406ce0ac82367afecc4a04R231

thegecko, 2018-07-24:
@viktornord An unexpected disconnect requires the device to have been connected in the first place.

As the [events being used](https://github.com/thegecko/webbluetooth/blob/master/src/adapter.ts#L229) are `once` events, by the time any diconnect happens the event listeners would have already been removed.

With the new changes, you need to call `disconnect()` in your loop before calling `connect()` again. The only other way around this would be to add a connection timeout, however this doesn't exist in the [webbluetooth specification](https://webbluetoothcg.github.io/web-bluetooth/). You can easily implement this now using the connect/disconnect methods available, if you wish.

PR #10 (2018-07-26): Ensure all event listeners are removed before connecting
Resolves #9
closed 2018-07-26 by thegecko