thegecko/webbluetooth #11: Accessing CSR8510 A10 device on Windows 10 fails, detected as poweredOff

YorVeX, 2019-03-27:
When I run noble/examples/advertisement-discovery.js it uses my CSR8510 A10 device (LED turns on), does some scanning and correctly finds the device I am trying to connect to (a Mi Band 2).

However, /webbluetooth/lib/adapter.js comes up with the error message "adapter not enabled". I removed the check for (this.state === true) check, because really all it does is hide the more useful accurate error message from the noble.startScanning() call.

**Now noble.startScanning() is failing with this error message:**
`Could not start scanning, state is poweredOff (not poweredOn)`

Also, the LED on the Bluetooth dongle doesn't light up so indeed it doesn't seem to be powered on.

Since the noble example script works fine maybe webbluetooth is just not using the noble library correctly? E.g. I found https://github.com/noble/noble/issues/228 where someone points out that per noble spec one should wait for the device to change the state to poweredOn. However, I don't have enough programming knowledge to know what to make of this.

Any ideas what I can try?

thegecko, 2019-05-27:
@YorVeX Any chance you were able to check the state?

YorVeX, 2019-05-28:
No, sorry, I've been quite busy during the last weeks. I will post here when I get the chance to check it.

YorVeX, 2019-06-25:
Actually the answer already lies in our discussion here. You said that possibly another state needs to be handled and I said I get an error message complaining that the state is "poweredOff".

So the solution was to change the code line to:
`if ((noble.state === "unknown") || (noble.state === "poweredOff"))`

Apparently the device just needs some time and noble.startScanning() was being called before it was ready. 

What also worked was to simply add some waiting time using setTimeout before the startScan call but since you already had a solution to wait for the state to become "poweredOn" using that is a lot cleaner.

Why doesn't the code just check whether the state is "poweredOn" and wait for state changes if it isn't, instead of only checking for "unknown" state? If I understand this correctly and state but "poweredOn" would be bad in this situation.

YorVeX, 2019-06-25:
So the good news is that this part now works, also the LED on the bluetooth dongle is lighting up.

The bad news is that it now fails with this error:
`Argh! getPrimaryService error: getPrimaryServices error: service not found`

Any ideas?
Should I open a separate issue for it?

thegecko, 2019-06-25:
Great, I'll add a fix for that state. Thanks for the info.

Regarding your other issue, does your connection to the device work in the browser?

YorVeX, 2019-06-25:
Forget the other issue. it was a mistake on my side. I had edited some stuff in the original miband.js file when hunting for the problem in the beginning (but that went in the wrong direction) and forgot to revert it. After reverting it everything works now and I get heart rate information from the Mi Band 2!

Thanks a lot! You can close this issue as soon as your fix is checked in.

thegecko, 2019-06-25:
sweet

PR #13 (2019-06-25): Add 'poweredOff' state check
Fixes #11 

cc @YorVeX
closed 2019-06-25 by thegecko