thegecko/webbluetooth #14: adapter not enabled (ASUS BT-400)

josemariagarcia95, 2019-07-12:
Hi.
I'm using the ASUS BT400 Bluetooth 4.0 adapter, and I have also installed the WinUSB driver with Zadig. I could put `noble `to work, but using this library only gets me an "adapter not enabled"

`requestDevice error: adapter not enabled`

**npm** version: v6.4.1
**node** version: v10.15.0

I'm working in **Windows 10 x64**, and I have installed **node-gyp**  and **node-bluetooth-hci-socket**. As I said, `noble `is working and detecting devices, but webbluetooth only gives me an `adapter not enabled`.

thegecko, 2019-08-20:
Please confirm the state of the adapter for your system in a similar way to the comments here: https://github.com/thegecko/webbluetooth/issues/11

thegecko, 2019-11-03:
Please re-open this issue with any further information

closed 2019-11-03 by thegecko