thegecko/webbluetooth #15: no devices found

playground, 2019-08-07:
With web bluetooth api, I was able to discover devices.  But with webbluetooth & node-web-bluetooth packages, it's getting no devices found...  What am I doing wrong?

```
bluetooth.requestDevice({acceptAllDevices: true})
.then(device => {
    console.log("connecting...");
    return device.gatt.connect();
})
.then(gattServer => {
    console.log("connected");
    server = gattServer;
    return enumerateGatt(server);
})
.then(() => server.disconnect())
.then(() => {
    console.log("\ndisconnected");
    process.exit();
})
.catch(logError);
```

thegecko, 2019-08-20:
Can you conform this device can be found using [noble](https://github.com/noble/noble)?

playground, 2019-08-20:
I had the same issue with noble, seems like it's broken for Mojave.

thegecko, 2019-08-20:
Oh that's not great :(
Could you raise an issue over there?

playground, 2019-08-29:
https://github.com/noble/noble/issues/874

c-reiter, 2019-09-13:
There should be a fix for the problems with noble on macos mojave. Just replace noble through @abandonware/noble (https://github.com/abandonware/noble). The version from abandonware should also work on windows/linux. Already replaced some other noble stuff through that and it worked perfectly.
@thegecko I tried replacing it in the webbluetooth package.json myself but then couldn't build webbluetooth, as there are some issues with node-gyp not finding some packages:
clang: error: no such file or directory: 'webbluetooth/node_modules/node-addon-api'
clang: error: no such file or directory: 'webbluetooth/node_modules/napi-thread-safe-callback'
Sadly I currently don't have time to try out fixing it myself, but if you could replace noble through @abandonware/noble, I would be glad to try if your fix works! Thanks for your amazing work on this package! :)

PR #16 (2019-11-05): Update deps to work with Node 12
Fixes #15
closed 2019-11-05 by thegecko