thegecko/webbluetooth #30: Can't write more than 20 bytes

Smart-XiaoLeiGe, 2021-01-14:
I write data as below, but can't get response. if i use a short data, will receive response.

await characteristicW.writeValueWithResponse(new Uint8Array([ 0x03, 0x04, 0x00, 0x1d, 0x40, 0xa2, 0x02, 0x01, 0x01, 0x6c, 0x65, 0x69, 0x2e, 0x77, 0x61, 0x6e, 0x67, 0x39, 0x40, 0x68, 0x33, 0x2e, 0x32, 0x2e, 0x31, 0x2e, 0x31, 0x36, 0x00, 0x05, 0x2a, 0xa9, 0x05 ]));

thegecko, 2021-01-14:
Things to try:

- Does this same code work in a browser?
- Does is succeed when using `writeValueWithoutResponse` ?

dlech, 2021-01-14:
It depends on both the OS and the device. Please see https://www.chromium.org/developers/how-tos/file-web-bluetooth-bugs for info on how to log Bluetooth packets to see what is really going on.

To be safe, you can limit data to 20 bytes per write and it should work on all devices as all devices have a minimum MTU of 23.

Smart-XiaoLeiGe, 2021-02-03:
> Things to try:
> 
> * Does this same code work in a browser?
> * Does is succeed when using `writeValueWithoutResponse` ?

Sorry for later response.
@thegecko 
- Does this same code work in a browser?
 Don't work in browser, i just use browser.
- Does is succeed when using writeValueWithoutResponse ?
NO

Smart-XiaoLeiGe, 2021-02-03:
> It depends on both the OS and the device. Please see https://www.chromium.org/developers/how-tos/file-web-bluetooth-bugs for info on how to log Bluetooth packets to see what is really going on.
>
> To [...]
Thank you for your guide.
But we need write long data, can we write more than 20 bytes by any method?

thegecko, 2021-02-03:
> Don't work in browser, i just use browser.

As this library is for implementing the WebBluetooth specification in a Node environment, I'd recommend you prove it works in a canonical browser environment first.

I'm closing this as it's not an issue specific to this library, but feel free to continue the discussion.

closed 2021-02-03 by thegecko