thegecko/webbluetooth #38: Ubuntu 18 compatibility ?

simonhill-fr, 2022-01-26:
Using webbluetooth with Node on the command line with macOS Mojave works flawlessly.

However, trying to port the same node script to Ubuntu, I manage to get a connection but I cannot subscribe to notifications, nor read data. Is this a known issue ?

thegecko, 2022-01-29:
I honestly can't remember and no longer have a linux machine to test with :(

The noble package offers a generic API for bluetooth, so I would expect if it works on one OS, it should work on all. If not, I suspect there may be an issue (or missing bit of setup) with the noble library.

There's some linux notes here:

https://github.com/abandonware/noble#linux

I also see there's a newer version you could try? `1.9.2-15`

Failing that, it may be worth trying to get noble working natively and raising any issue there?

simonhillfr, 2022-02-21:
Thanks, I ended up using the browser webbluetooth but it's not perfect so I'll have a look into noble

closed 2022-02-21 by thegecko