thegecko/webbluetooth #43: Please release new package version to npm, containing latest merged changes

noisy, 2022-09-30:
I've noticed that https://www.npmjs.com/package/webbluetooth/ was last time published over 2 years ago. A lot of people may not notice `next` version, and because of that they may think this library was not maintain for last 2 years.

Also, latest next version do not contain latest changes, like those: https://github.com/thegecko/webbluetooth/pull/42

I would greatly appreciate a new release :)

BTW, great job with this library! :) Thanks to it, I'm going to be able to write an integration of my smart-toothbrush with my smart-mirror :)

thegecko, 2022-09-30:
Untested:

https://www.npmjs.com/package/webbluetooth/v/2.2.0

>  I'm going to be able to write an integration of my smart-toothbrush with my smart-mirror

:/

closed 2022-09-30 by thegecko

noisy, 2022-10-03:
> https://www.npmjs.com/package/webbluetooth/v/2.2.0

great! Thanks!

thegecko, 2022-10-03:
> I'm going to be able to write an integration of my smart-toothbrush with my smart-mirror

Video evidence required!