thegecko/webbluetooth #77: Error: Write failed on every attempt

Supermrk, 2023-07-05:
Whenever I try to write to a valid characteristic, the error "Write failed" always occurs. I'm trying to implement some code I found in the https://github.com/ib0b/RGB-PC repo into a NodeJS webserver project. The RGB-PC project worked perfectly fine on my machine, but implementing the same code with the webbluetooth library causes this error to happen.

```js
            this.server = await device.gatt.connect();
            // await this.server.connect();

            this.service = await this.server.getPrimaryService(
                "00010203-0405-0607-0809-0a0b0c0d1910"
            );
            this.characteristic = await this.service.getCharacteristic(
                "00010203-0405-0607-0809-0a0b0c0d2b11"
            );

            await this.characteristic.writeValue(commands.get("turnOn"));
            
            let commands = {
                "turnOn": "3301010000000000000000000000000000000033",
                "turnOff": "3301000000000000000000000000000000000032",
                "keepAlive": "aa010000000000000000000000000000000000ab",
                //music
                "energetic": "3305130563000000000000000000000000000043",
                "spectrum": "3305130463000000000000000000000000000042",
                "rythm": "3305130363000000000000000000000000000045",
                "separation": "3305133263000000000000000000000000000074",
                "rolling": "3305130663000000000000000000000000000040",

                get(command) {
                    let hex = this[command];

                    return new Uint8Array(
                        hex.match(/[\da-f]{2}/gi).map(function (h) {
                           return parseInt(h, 16);
                        })
                    );

                },
                convert(string) {
                    return new Uint8Array(
                        string.match(/[\da-f]{2}/gi).map(function (h) {
                            return parseInt(h, 16);
                        })
                    );
                }

            }
```

thegecko, 2023-08-22:
This may be the issue:

https://github.com/thegecko/webbluetooth/issues/89

lgazo, 2024-03-05:
I face similar issue when using [muse-js](https://github.com/urish/muse-js/) with webbluetooth 3.2.1. I bumped the version from 3.0.1 as I thought that might resolve the issue. Unfortunately it did not. The `muse-js` works with my device when used with Browser's Web Bluetooth - e.g. with https://github.com/kylemath/EEGEdu project.

This is what I get when I call `await muse_client.start();` that uses `webbluetooth` lib:

```
/home/user/development/muse-collector/node_modules/.pnpm/webbluetooth@3.2.1/node_modules/webbluetooth/dist/adapters/simpleble-adapter.js:468
                    throw new Error('Write failed');
                          ^

Error: Write failed
    at SimplebleAdapter.<anonymous> (/home/user/development/muse-collector/node_modules/.pnpm/webbluetooth@3.2.1/node_modules/webbluetooth/dist/adapters/simpleble-adapter
.js:468:27)
    at step (/home/user/development/muse-collector/node_modules/.pnpm/webbluetooth@3.2.1/node_modules/webbluetooth/dist/adapters/simpleble-adapter.js:72:23)
    at Object.next (/home/user/development/muse-collector/node_modules/.pnpm/webbluetooth@3.2.1/node_modules/webbluetooth/dist/adapters/simpleble-adapter.js:53:53)
    at /home/user/development/muse-collector/node_modules/.pnpm/webbluetooth@3.2.1/node_modules/webbluetooth/dist/adapters/simpleble-adapter.js:47:71
    at new Promise (<anonymous>)
    at __awaiter (/home/user/development/muse-collector/node_modules/.pnpm/webbluetooth@3.2.1/node_modules/webbluetooth/dist/adapters/simpleble-adapter.js:43:12)
    at SimplebleAdapter.writeCharacteristic (/home/user/development/muse-collector/node_modules/.pnpm/webbluetooth@3.2.1/node_modules/webbluetooth/dist/adapters/simpleble
-adapter.js:455:16)
    at BluetoothRemoteGATTCharacteristicImpl.<anonymous> (/home/user/development/muse-collector/node_modules/.pnpm/webbluetooth@3.2.1/node_modules/webbluetooth/dist/chara
cteristic.js:250:65)
    at step (/home/user/development/muse-collector/node_modules/.pnpm/webbluetooth@3.2.1/node_modules/webbluetooth/dist/characteristic.js:72:23)
    at Object.next (/home/user/development/muse-collector/node_modules/.pnpm/webbluetooth@3.2.1/node_modules/webbluetooth/dist/characteristic.js:53:53)

Node.js v20.11.1
```

thegecko, 2025-02-01:
Please let me know if your code works correctly in chrome

Sfinx, 2025-05-23:
Any ideas how to catch this "throw new Error('Write failed')" ?