thegecko/webbluetooth #124: Does requestDevice() find devices by service UUID?

cwilling, 2023-11-18:
I have some devices which advertise a service using a 128bit UUID generated at [https://www.uuidgenerator.net/](https://www.uuidgenerator.net/). The devices are successfully discovered in a browser using my own code based on webbluetooth samples at [https://googlechrome.github.io/samples/web-bluetooth/index.html](https://googlechrome.github.io/samples/web-bluetooth/index.html) e.g.
```
navigator.bluetooth.requestDevice(options)
```
(where options are: `{"filters":[{"services":["9f9c635f-3321-4490-9bae-c58572125000"]}]}`)

However if I try the  heartrate.js example here but change the 'heart_rate' filter to my 128bit uuid i.e. change `filters: [{ services: [ "heart_rate" ] }]` to `filters: [{ services: [ "9f9c635f-3321-4490-9bae-c58572125000" ] }]`, the devices are not discovered.

Should I be able to use arbitrary 128bit service uuids to filter available devices?

P.S. the devices are all found, along with a bunch or other devices, using the list.js example. I just want to limit the listing to those devices with my own service uuid.

Thanks for any assistance.

thegecko, 2023-11-18:
Thanks for raising this @cwilling 

Can you confirm if your service UUID is advertised by your device (not just a primary service)?

This library currently only matches service UUID filters against advertised services and I've just tested this works.
However, the specification is confusing and could be seen as contradicting itself around this. So the implementation may be incorrect either in chrome or this library!

e.g.

https://webbluetoothcg.github.io/web-bluetooth/#scan-for-devices

> 6. / 2. If device’s advertised [Service UUIDs](https://webbluetoothcg.github.io/web-bluetooth/#service-uuid-data-type) have a non-empty intersection with the set of [Service](https://webbluetoothcg.github.io/web-bluetooth/#service) UUIDs, add device to result and abort these sub-steps.

and:

https://webbluetoothcg.github.io/web-bluetooth/#matches-a-filter

> 3. For each uuid in filter.[services](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothlescanfilterinit-services), if the UA has not received advertising data, an [extended inquiry response](https://webbluetoothcg.github.io/web-bluetooth/#extended-inquiry-response), or a service discovery response indicating that the device supports a primary (vs included) service with UUID uuid, return mismatch.

@reillyeon What is the intent here? Should serviceuuid filters match advertised services and primary services?

cwilling, 2023-11-19:
Thanks for looking into this.

I can confirm the correct service UUID is being advertised (using Light Blue app on my phone).

Because my peripheral  devices offer multiple services of which different clients may use different subsets, I use a separate UUID just for advertising. That way a client finding a device with correct UUID will know what real services  the device offers (or connect to confirm). This advertising UUID has no corresponding service - it's purely for advertising. However from your discussion above, it seems expected that the advertised UUID also corresponds to an actual service. To test whether that makes a difference, I have now added a primary service with the same UUID as the advertising UUID. Unfortunately it has made no difference.

Other information:
- for the ESP32 based peripheral devices, I use the NimBLE-Arduino library
- I also have equivalent python code running on a Raspberry Pi Zero W

They're all advertising using the same UUID and are successfully filtered by that UUID using the browser application.

thegecko, 2023-11-19:
Can you run the select example:

```bash
> node examples/selector.js
```

This should display all devices in range and any services they advertise. e.g. I see:

```
scanning...
select a device:
1: BBC micro:bit [pezut]
2: Garagentorantrieb
	Advertising: 669a9001-0008-968f-e311-6050405558b3
```

cwilling, 2023-11-19:
I previously copied examples/selector.js into the directory from where I want to run things (when they're working). Therefore I changed the line `const Bluetooth = require("../").Bluetooth;` to `const Bluetooth = require('webbluetooth').Bluetooth;`

```
chris@d8:~/src/dcdp-server/scripts/net$ node ./selector.js 
scanning...
select a device:
1: esp32-a0b765dcbdce
2: Unknown or Unsupported Device (EC:81:93:A6:60:6A)
3: LE-Indi Bose Micro
4: Unknown or Unsupported Device (68:C8:D1:6D:B9:52)
5: Unknown or Unsupported Device (45:79:A8:15:3E:1C)
6: Unknown or Unsupported Device (88:C6:26:CF:BC:F9)
7: esp32-ec6260bcaec2
8: esp32-348518438aa1
9: Unknown or Unsupported Device (48:F9:F3:EA:1D:07)
10: Unknown or Unsupported Device (46:D3:00:18:A9:17)
11: Unknown or Unsupported Device (6E:22:61:A6:85:F4)
12: Unknown or Unsupported Device (44:C6:5D:84:78:45)
13: Unknown or Unsupported Device (5D:37:A0:F3:F0:A4)
14: Unknown or Unsupported Device (46:6A:33:B2:F7:F4)
15: Unknown or Unsupported Device (74:13:1C:E5:D7:BD)
16: Unknown or Unsupported Device (6D:9A:2B:70:6A:E9)
17: Unknown or Unsupported Device (3F:5E:6C:80:76:66)
18: Unknown or Unsupported Device (56:92:E5:6F:AD:1F)
19: Unknown or Unsupported Device (68:00:60:51:EB:22)
20: Unknown or Unsupported Device (60:75:C3:16:4C:18)
21: Unknown or Unsupported Device (61:B5:AF:7E:C8:FA)
22: Unknown or Unsupported Device (74:92:70:DA:F7:70)
23: Unknown or Unsupported Device (65:0F:B6:64:95:78)
24: Unknown or Unsupported Device (49:B5:77:E4:EB:9C)
25: Unknown or Unsupported Device (70:29:BA:17:E5:C2)
26: Unknown or Unsupported Device (78:3D:7F:EA:FC:12)
27: Unknown or Unsupported Device (43:73:20:C3:87:E9)
```
If I then select device 1, I see all the three primary services including the 9f9c635f-3321-4490-9bae-c58572125000 service which is UUID being advertised:
```
connecting...
connected
service: 00001801-0000-1000-8000-00805f9b34fb
        └characteristic: 00002a05-0000-1000-8000-00805f9b34fb
                └descriptor: 00002902-0000-1000-8000-00805f9b34fb
service: 9f9c635f-3321-4490-9bae-c58572125000
service: 9f9c635f-3321-4490-9bae-c58572125001
        └characteristic: 9f9c635f-3321-4490-9bae-c58572125002
                └descriptor: 00002902-0000-1000-8000-00805f9b34fb
        └characteristic: 9f9c635f-3321-4490-9bae-c58572125003
                └descriptor: 00002902-0000-1000-8000-00805f9b34fb
        └characteristic: 9f9c635f-3321-4490-9bae-c58572125006
                └descriptor: 00002902-0000-1000-8000-00805f9b34fb
        └characteristic: 9f9c635f-3321-4490-9bae-c58572125007
                └descriptor: 00002902-0000-1000-8000-00805f9b34fb
        └characteristic: 9f9c635f-3321-4490-9bae-c58572125008
                └descriptor: 00002902-0000-1000-8000-00805f9b34fb
        └characteristic: 9f9c635f-3321-4490-9bae-c58572125009
                └descriptor: 00002902-0000-1000-8000-00805f9b34fb
        └characteristic: 9f9c635f-3321-4490-9bae-c58572125010
                └descriptor: 00002902-0000-1000-8000-00805f9b34fb
        └characteristic: 9f9c635f-3321-4490-9bae-c58572125011
                └descriptor: 00002902-0000-1000-8000-00805f9b34fb
        └characteristic: 9f9c635f-3321-4490-9bae-c58572125012
                └descriptor: 00002902-0000-1000-8000-00805f9b34fb
        └characteristic: 9f9c635f-3321-4490-9bae-c58572125014
                └descriptor: 00002902-0000-1000-8000-00805f9b34fb
        └characteristic: 9f9c635f-3321-4490-9bae-c58572125015
                └descriptor: 00002902-0000-1000-8000-00805f9b34fb
service: 9f9c635f-3321-4490-9bae-c58572125100
        └characteristic: 9f9c635f-3321-4490-9bae-c58572125101
                └descriptor: 00002902-0000-1000-8000-00805f9b34fb
        └characteristic: 9f9c635f-3321-4490-9bae-c58572125102
                └descriptor: 00002902-0000-1000-8000-00805f9b34fb

disconnected
```
That all lines up with how the services have been setup for the peripherals.

Interestingly (not sure whether related to the main problem), I just noticed that the Raspberry Pi devices isn't listed even though it's found OK by the browser app.

cwilling, 2023-11-20:
So far, I have been trying with Linux.

I have access to a mac laptop which I've just tried. For all the esp32 & raspi devices, selector.js shows additional output "Advertising: " + the advertising UUID. Also, when I substitute that UUID into example/heartrate.js (replacing "heart_rate" filter), a relevant device is found i.e. filtering by UUID works as "advertised" on Mac but not Linux.

If I now need to investigate SimpleBLE, do you have any tips as to where to start looking?

thegecko, 2023-11-20:
OK, this sounds like an OS specific issue with SimpleBLE.

@Symbitic can you support with investigating this shortcoming?

Symbitic, 2023-11-21:
@thegecko Yes, this is an issue with SimpleBLE. More specifically, a problem with SimpleBlueZ: OpenBluetoothToolbox/SimpleBLE#135
I'm trying to look into it when I have time. One thing to help in the meantime: Service data before connecting doesn't work on Linux yet, but Machine data works on all platforms.

cwilling, 2023-11-21:
Thanks @Symbitic that's great news - can't wait to try it out!

thegecko, 2023-11-22:
A great example of people working together (and making my life easier)! thanks both.

thegecko, 2023-11-22:
Will update webbluetooth when this is merged: https://github.com/OpenBluetoothToolbox/SimpleBLE/pull/275

Symbitic, 2023-11-22:
Great! Any chance you can take a look at my PR while we wait?

cwilling, 2023-12-01:
@Symbitic's  PR has been merged!

thegecko, 2023-12-01:
The update run is here (it built):
https://github.com/thegecko/webbluetooth/actions/runs/7064331341

Perhaps use the artifacts on that to see if it fixes the issue?

cwilling, 2023-12-02:
I was able to pull from your repo and checkout the advertised-uuids branch and build a new build/Release/simpleble.node that works.

However I couldn't use the simpleble.node from the artifacts. Trying to use it complains about not finding a particular glibc version. I had the same problem with the previous version (before rebuilding) so since this isn't related to the original issue, I may raise a separate issue about it.

closed 2023-12-02 by thegecko

thegecko, 2024-02-12:
Annoyingly I can't use the latest version of SimpleBLE with the advert fix due to this bug being introduced:

https://github.com/OpenBluetoothToolbox/SimpleBLE/issues/277