thegecko/webbluetooth #128: Setting scanTime

cwilling, 2023-11-26:
I've tried adding a scanTime to options in both heartrate & selector examples but it doesn't seem to have any effect. e.g.
```
    const device = await bluetooth.requestDevice({
             scanTime:       20,
             filters: [{ services: [ "heart_rate" ] }]
    });
```
Is that the right way to do it? It has no effect - still times out after default ~10 seconds.

Thanks for any tips,
chris

thegecko, 2023-11-27:
Hi @cwilling 

`scanTime` is an option on the extended bluetooth options and not part of the specification:

https://github.com/thegecko/webbluetooth/blob/master/src/bluetooth.ts#L34

To use it, create a new Bluetooth instance, e.g.:

```
const bluetooth = new Bluetooth({
    scanTime: 20
});

const device = await bluetooth.requestDevice({
    filters: [{ services: [ "heart_rate" ] }]
});
```

cwilling, 2023-11-28:
Thanks, that works.

I should have seen it was one of BluetoothOptions for the Bluetooth constructor, nothing to do with filters for requestDevice().

closed 2023-11-28 by cwilling