thegecko/webbluetooth #132: Compatibility of build/Release/simpleble.node

cwilling, 2023-12-02:
When running any example, the supplied simpleble.node generates an error:
```
chris@d8:~/src/webbluetooth$ node examples/selector.js 
/home/chris/src/webbluetooth/node_modules/bindings/bindings.js:121
        throw e;
        ^

Error: /lib64/libc.so.6: version `GLIBC_2.34' not found (required by /home/chris/src/webbluetooth/build/Release/simpleble.node)
    at Object.Module._extensions..node (node:internal/modules/cjs/loader:1189:18)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at bindings (/home/chris/src/webbluetooth/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/home/chris/src/webbluetooth/dist/adapters/simpleble.js:29:36)
    at Module._compile (node:internal/modules/cjs/loader:1105:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1159:10)
    at Module.load (node:internal/modules/cjs/loader:981:32) {
  code: 'ERR_DLOPEN_FAILED'
}

```
I can fix this locally by running `npm run build:all` which generates a new simpleble.node and everything works fine.

I initially thought the error would be due to your build machine having a glibc version which is too recent for what I have on the machine I'm using. I see in your .github/workflows/prebuild.yml that linux-x64 jobs run on ubuntu-latest and maybe that is a problem for other systems out in the wild (not everyone wants/is able to run the very latest). However I also tried all this on another machine running Ubuntu 22.04 (full updated) and had the same issue so I'm a bit confused.

thegecko, 2023-12-02:
@cwilling to help me tie down the issue, can you confirm if this is broken in the released versions? I know you were testing with some unreleased binaries.

cwilling, 2023-12-02:
However back on my normal machine with glibc-2.33, even the newer artifacts version fails because it wants glibc-2.34. In that case the only solution is to rebuild simpleble.node locally.

I'm not sure there's anything you can do about that - probably a result of ubunt-latest in your workflow.

thegecko, 2023-12-02:
OK, I think the best fix for this will be to use `prebuildify-cross` in a similar way to [how we do it in node-usb](https://github.com/node-usb/node-usb/blob/master/.github/workflows/prebuild.yml#L33).
This will require [prebuildify to support cmake-js](https://github.com/prebuild/prebuildify/issues/49), which I'll look in to.

thegecko, 2025-03-09:
v3.3.2 now ships with binaries included

closed 2025-03-09 by thegecko