thegecko/webbluetooth #177: addEventHandler should accept "null" to remove event handlers as per W3C specs

Apollon77, 2024-03-22:
see https://html.spec.whatwg.org/multipage/webappapis.html#event-handler-attributes

> a listener, which is either null or an [event listener](https://dom.spec.whatwg.org/#concept-event-listener) responsible for running [the event handler processing algorithm](https://html.spec.whatwg.org/multipage/webappapis.html#the-event-handler-processing-algorithm). Initially, an [event handler](https://html.spec.whatwg.org/multipage/webappapis.html#event-handlers)'s [listener](https://html.spec.whatwg.org/multipage/webappapis.html#event-handler-listener) must be set to null.

I can provide a PR once #56 is in if wanted

PS: I curently work on https://github.com/project-chip/matter.js and want to get a web-bluetooth compatible  BLE handler for Matter commissioning, so that developers can choose if they use it in a browser or node.js :-)

Apollon77, 2024-03-22:
@thegecko i can also do a PR then for the types package to get tthat in sync because it is missing there too

thegecko, 2024-04-13:
Yes please