thegecko/webbluetooth #179: Build error on raspberry 4

Apollon77, 2024-03-23:
When using webbluetooth I get (after installing cmake):

```
ingof@raspi16g:~/official-matter.js$ npm i
npm ERR! code 1
npm ERR! path /home/ingof/official-matter.js/node_modules/webbluetooth
npm ERR! command failed
npm ERR! command sh -c prebuild-install --backend cmake-js --runtime napi || cmake-js rebuild
npm ERR! Not searching for unused variables given on the command line.
npm ERR! -- The C compiler identification is GNU 11.4.0
npm ERR! -- The CXX compiler identification is GNU 11.4.0
npm ERR! -- Detecting C compiler ABI info
npm ERR! -- Detecting C compiler ABI info - done
npm ERR! -- Check for working C compiler: /usr/bin/cc - skipped
npm ERR! -- Detecting C compile features
npm ERR! -- Detecting C compile features - done
npm ERR! -- Detecting CXX compiler ABI info
npm ERR! -- Detecting CXX compiler ABI info - done
npm ERR! -- Check for working CXX compiler: /usr/bin/c++ - skipped
npm ERR! -- Detecting CXX compile features
npm ERR! -- Detecting CXX compile features - done
npm ERR! -- Configuring incomplete, errors occurred!
npm ERR! See also "/home/ingof/official-matter.js/node_modules/webbluetooth/build/CMakeFiles/CMakeOutput.log".
npm ERR! prebuild-install warn install No prebuilt binaries found (target=6 runtime=napi arch=arm64 libc= platform=linux)
npm ERR! info TOOL Using Unix Makefiles generator.
npm ERR! info CMD CLEAN
npm ERR! info RUN [
npm ERR! info RUN   'cmake',
npm ERR! info RUN   '-E',
npm ERR! info RUN   'remove_directory',
npm ERR! info RUN   '/home/ingof/official-matter.js/node_modules/webbluetooth/build'
npm ERR! info RUN ]
npm ERR! info CMD CONFIGURE
npm ERR! info RUN [
npm ERR! info RUN   'cmake',
npm ERR! info RUN   '/home/ingof/official-matter.js/node_modules/webbluetooth',
npm ERR! info RUN   '--no-warn-unused-cli',
npm ERR! info RUN   '-G',
npm ERR! info RUN   'Unix Makefiles',
npm ERR! info RUN   '-DCMAKE_JS_VERSION=7.3.0',
npm ERR! info RUN   '-DCMAKE_BUILD_TYPE=Release',
npm ERR! info RUN   '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/home/ingof/official-matter.js/node_modules/webbluetooth/build/Release',
npm ERR! info RUN   '-DCMAKE_MSVC_RUNTIME_LIBRARY=MultiThreaded$<$<CONFIG:Debug>:Debug>',
npm ERR! info RUN   '-DCMAKE_JS_INC=/home/ingof/official-matter.js/node_modules/node-api-headers/include;/home/ingof/official-matter.js/node_modules/webbluetooth/node_modules/node-addon-api',
npm ERR! info RUN   '-DCMAKE_JS_SRC=',
npm ERR! info RUN   '-DNODE_RUNTIME=node',
npm ERR! info RUN   '-DNODE_RUNTIMEVERSION=18.17.1',
npm ERR! info RUN   '-DNODE_ARCH=arm64',
npm ERR! info RUN   '-DCMAKE_JS_LIB=',
npm ERR! info RUN   '-DCMAKE_CXX_FLAGS=-DBUILDING_NODE_EXTENSION'
npm ERR! info RUN ]
npm ERR! CMake Error at CMakeLists.txt:14 (add_subdirectory):
npm ERR!   add_subdirectory given source "SimpleBLE/simpleble" which is not an
npm ERR!   existing directory.
npm ERR! 
npm ERR! 
npm ERR! ERR! OMG Process terminated: 1

npm ERR! A complete log of this run can be found in: /home/ingof/.npm/_logs/2024-03-23T16_03_00_266Z-debug-0.log
```

It seems that SimpleBLE is not delivered as source files in the npm package -.. which is true ... https://github.com/thegecko/webbluetooth/blob/master/.npmignore#L4

SHould it be added to allow fallback local builds?

thegecko, 2024-03-25:
> SHould it be added to allow fallback local builds?

Hmm, good question. We should either ship the source or a prebuilt binary for your scenario. We currently do neither.

I could add the missing prebuild targets similar to https://github.com/node-usb/node-usb/blob/master/.github/workflows/prebuild.yml#L18 which would be the best long term fix, but short-term would you mind crafting a PR to modify the `.vscodeignore` file to ship exactly what you need to build locally?

Apollon77, 2024-03-25:
Ideally both :-) So shiop soiurce that - for the special cases you can not prebuild - always a local build works ... and yes you could add common sources like armhf, armv7 and such targets too. But that could never be really complete, so I would go both ways. Yes I can do a PR

FirokOtaku, 2024-04-03:
I get the same error when trying to run `npm i webbluetooth` on a X64 Windows machine. 🤣

thegecko, 2024-04-13:
> I get the same error when trying to run npm i webbluetooth on a X64 Windows machine. 🤣

Which architecture do you have?

FirokOtaku, 2024-04-13:
> > I get the same error when trying to run npm i webbluetooth on a X64 Windows machine. 🤣
> 
> Which architecture do you have?

All my machines are x86_64 based.

thegecko, 2024-04-13:
> All my machines are x86_64 based.

That should be fine, we bundle the native binary for that. Can you share any further logs?

FirokOtaku, 2024-04-16:
> That should be fine, we bundle the native binary for that. Can you share any further logs?

Sure.

----

Commands used:

```bash
cd /d "T:\Projects\web-bluetooth-test"
npm install webbluetooth
```

----

<details>
<summary>Short log in command line</summary>

```log
npm ERR! code 1
npm ERR! path T:\Projects\web-bluetooth-test\node_modules\webbluetooth
npm ERR! command failed
npm ERR! command C:\Windows\system32\cmd.exe /d /s /c prebuild-install --backend cmake-js --runtime napi || cmake-js rebuild
npm ERR! Not searching for unused variables given on the command line.
npm ERR! -- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.19045.
npm ERR! -- The C compiler identification is MSVC 19.38.33133.0
npm ERR! -- The CXX compiler identification is MSVC 19.38.33133.0
npm ERR! -- Detecting C compiler ABI info
npm ERR! -- Detecting C compiler ABI info - done
npm ERR! -- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.38.33130/bin/Hostx64/x64/cl.exe - skipped
npm ERR! -- Detecting C compile features
npm ERR! -- Detecting C compile features - done
npm ERR! -- Detecting CXX compiler ABI info
npm ERR! -- Detecting CXX compiler ABI info - done
npm ERR! -- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.38.33130/bin/Hostx64/x64/cl.exe - skipped
npm ERR! -- Detecting CXX compile features
npm ERR! -- Detecting CXX compile features - done
npm ERR! Microsoft (R) Library Manager Version 14.38.33133.0
npm ERR! Copyright (C) Microsoft Corporation.  All rights reserved.
npm ERR!
npm ERR!   正在创建库 T:\Projects\web-bluetooth-test\node_modules\webbluetooth\build\node.lib 和对象 T:\Projects\web-bluetooth-test\node_modules\webbluetooth\build\node.exp
npm ERR! -- Configuring incomplete, errors occurred!
npm ERR! prebuild-install warn install aborted
npm ERR! info find VS using VS2022 (17.8.34330.188) found at:
npm ERR! info find VS "C:\Program Files\Microsoft Visual Studio\2022\Community"
npm ERR! info find VS run with --verbose for detailed information
npm ERR! info TOOL Using Visual Studio 17 2022 generator.
npm ERR! info CMD CLEAN
npm ERR! info RUN [
npm ERR! info RUN   'cmake',
npm ERR! info RUN   '-E',
npm ERR! info RUN   'remove_directory',
npm ERR! info RUN   'T:\\Projects\\web-bluetooth-test\\node_modules\\webbluetooth\\build'
npm ERR! info RUN ]
npm ERR! info CMD CONFIGURE
npm ERR! info find VS using VS2022 (17.8.34330.188) found at:
npm ERR! info find VS "C:\Program Files\Microsoft Visual Studio\2022\Community"
npm ERR! info find VS run with --verbose for detailed information
npm ERR! info RUN [
npm ERR! info RUN   'cmake',
npm ERR! info RUN   'T:\\Projects\\web-bluetooth-test\\node_modules\\webbluetooth',
npm ERR! info RUN   '--no-warn-unused-cli',
npm ERR! info RUN   '-G',
npm ERR! info RUN   'Visual Studio 17 2022',
npm ERR! info RUN   '-A',
npm ERR! info RUN   'x64',
npm ERR! info RUN   '-DCMAKE_JS_VERSION=7.3.0',
npm ERR! info RUN   '-DCMAKE_BUILD_TYPE=Release',
npm ERR! info RUN   '-DCMAKE_RUNTIME_OUTPUT_DIRECTORY=T:\\Projects\\web-bluetooth-test\\node_modules\\webbluetooth\\build',
npm ERR! info RUN   '-DCMAKE_MSVC_RUNTIME_LIBRARY=MultiThreaded$<$<CONFIG:Debug>:Debug>',
[... 657 lines cut ...]
567 error
567 error ERR! OMG Process terminated: 1
568 verbose exit 1
569 timing npm Completed in 43736ms
570 verbose unfinished npm timer reify 1713231020605
571 verbose unfinished npm timer reify:build 1713231037405
572 verbose unfinished npm timer build 1713231037406
573 verbose unfinished npm timer build:deps 1713231037406
574 verbose unfinished npm timer build:run:install 1713231037419
575 verbose unfinished npm timer build:run:install:node_modules/webbluetooth 1713231037419
576 verbose code 1
577 error A complete log of this run can be found in: C:\Users\firok\AppData\Local\npm-cache\_logs\2024-04-16T01_30_15_193Z-debug-0.log
```

</details>

----

Enviroment information:

* cmake 3.29.0
* npm 9.6.7
* node 20.3.1
* Visual Studio Comunity 2022
* Windows 10 22H2 19045.4291

thegecko, 2024-04-29:
Thanks for the logs, the install should be getting the prebuild binary for your platform rather than running a build

thegecko, 2024-04-29:
@FirokOtaku I've just tested this on x86_64 Windows 10 machine and it worked as expected.

`npm install webbluetooth` downloaded a prebuild for the host rather than running the build process.

npm: 10.x
node: 20.x

This may be an issue with your specific machine, but #180 will now allow builds to run locally anyway (if all prerequisites are met).

FirokOtaku, 2024-04-29:
@thegecko It's wired that I just tried `npm i webbluetooth` on my home PC (with the same architecture of x86_64 windows. Only different is it runs with NodeJS 19) and it runs successfully.

But it still not works on my work PC (which generated the error logs above).

So, yeah, that seems an machine-specific issue. 🤣 Maybe I should wipe some environment variables or have a full clean install of NodeJS.

Apollon77, 2024-04-29:
@thegecko One question ... when looking at https://github.com/thegecko/webbluetooth/blob/master/.github/workflows/prebuild.yml#L42 I could guess that prebuild is only done for nodejs 16 ?!  SO not 18, not 20 ... (ok also not 19) ... but ok I do not know example when N-ABI potentially changed or was stabkle, but could it be taht 16..19 had the same ABI and Node.js 20 has a different one? (just guessing). 

And yes without #180 any local build will fail

thegecko, 2024-04-29:
We use NAPI which is backwards compatible and not tied to Node versions:

NAPI 6: https://github.com/thegecko/webbluetooth/blob/master/package.json#L55

Which should be v10.20.0+, v12.17.0+, 14.0.0 and all later versions: https://nodejs.org/docs/v20.12.1/api/n-api.html#node-api-version-matrix

thegecko, 2024-04-29:
@FirokOtaku I'm going to close this issue as it seems to be setup specific and a little OT :) Feel free to open a new issue if you get to the bottom of it.

PR #180 (2024-04-29): Allow local build
This PR adds the SimpleBLE directory and all sub dirs to the npm package. I verified with publish dry run" that it would be added to the package and validated that it can install on a raspi 4 with an "npm pack" version of that.

BTW: A GitHub install would _not_ check the SimpleBLE directory out and so would not work here!

Means: I could test it only with npm pack

fixes #179 

PS: The PR contains two extra commits from trying to get more debug infos, just squash&merge it ...
closed 2024-04-29 by thegecko