thegecko/webbluetooth #190: Add prebuilds for arm* architectures

Apollon77, 2024-04-20:
A build on Raspi 4 takes 12 minutes, I did not tried Raspi 3 :-)

So yes adding the usual arm architectures (arm64, armv7, armv6) - maybe see relevant list from a noble repo at https://github.com/stoprocent/node-bluetooth-hci-socket/blob/main/.github/workflows/build.yml#L14 might be a good idea

Apollon77, 2024-04-20:
Ohh I see that it was started with https://github.com/thegecko/webbluetooth/pull/133/files already

thegecko, 2024-04-20:
> Ohh I see that it was started with https://github.com/thegecko/webbluetooth/pull/133/files already

Yeah, I couldn't get prebuildify to work with cmake-js which would prebuild a lot more binaries and include them with the package. Feel free to continue my work.

The setup on [node-usb](https://github.com/node-usb/node-usb/blob/master/.github/workflows/prebuild.yml) may help

Apollon77, 2024-04-20:
Ok, I had a look ... so in fact for prebuildify it would be needed to use own docker images or to add cmake to the "prebuild" docker images.

thegecko, 2024-04-29:
I use custom docker images for node-usb, see: https://github.com/node-usb/docker-images.

Specifially: https://github.com/node-usb/docker-images/blob/main/linux-arm64/Dockerfile

Apollon77, 2024-04-29:
Would it make sense to add cmake there too to share those or do we need another repo for having cmake available? Then I could only provide them as "personal repo" unless you would like to add them here ...

thegecko, 2024-04-29:
Hmm, we could add cmake there. If we need to split them in future, we can then worry about separate repos

Apollon77, 2024-04-29:
Do you want to do that or should I try to add a PR if I find time and get that tackled somehow? ;-) (my Docker knowledge is more limited :-)

thegecko, 2024-04-29:
I have little time. Can you try to get a prebuild flow setup using those containers and we can add tools as needed.

Apollon77, 2024-04-29:
No promises currently. Also my time is limited and that task is far away from my comfort and knowledge zone tbh ... so I most likely will need a lot of time ... let's see

thegecko, 2025-03-09:
v3.3.2 now includes bundled binaries including the arm version for Mac and Linux

closed 2025-03-09 by thegecko