thegecko/webbluetooth #191: macOS do not work

Apollon77, 2024-04-20:
I tried the lib on my macos Sonoma 14.4.1 (23E224) and it do not scan/find any devices and also do not expose the hardware adapter list (see #189).

There are no building error when building manually on my host and also no usage errors. I tried with root rights and without nd yes also my terminal app has proper access rights for bluetooth as well as the nodejs binary.

Apollon77, 2024-04-22:
@thegecko Maybe upgrading SimpleBLE would be an idea? Currently webbluetooth bases on last 0.6.x of SimpleBLE. What about upgrding to current 0.7.3? Especially for macos there were big changes.

I could try to make a PR and test before ... I just need to find out how to update the "repo link"

thegecko, 2024-04-22:
I develop this library on MacOS Sonoma, so pretty sure it works! Are you on intel or Arm architecture?

Apollon77, 2024-04-22:
Wtf ... I was 1000% sure last days it do not work ... now it works with 0.6.1 ... also returning adapter list works with my PR ... what the ... grrmmppff ... Sorry

Apollon77, 2024-04-22:
PS: But executing test uite locally (intel Mac) still gives errors:

```
npm run test                  

> webbluetooth@3.2.1 test
> mocha --timeout 10000 test/*.test.js

  bluetooth module
    ✔ should describe basic constants

  device
    1) "before each" hook for "should return a device with requestDevice"

  devices
    ✔ should return no devices with getDevices (2067ms)
    2) should return one device with getDevices after a request
    3) should forget devices

  all devices
    ✔ should return all devices with getDevices (2073ms)

  services
    4) "before each" hook for "should get services"
    5) "after each" hook for "should get services"

  characteristics
    6) "before each" hook for "should get characteristics"
    7) "after each" hook for "should get characteristics"

  descriptors
    8) "before each" hook for "should get descriptors"
    9) "after each" hook for "should get descriptors"

  bluetoothuuid
    ✔ should get service
    ✔ should get characteristic
    ✔ should get descriptor
    ✔ should get canonical uuid

  7 passing (29s)
  9 failing

  1) device
       "before each" hook for "should return a device with requestDevice":
     Error: the string "requestDevice error: no devices found" was thrown, throw an Error :)
      at runNextTicks (node:internal/process/task_queues:60:5)
      at process.processTimers (node:internal/timers:511:9)

  2) devices
       should return one device with getDevices after a request:
     Error: the string "requestDevice error: no devices found" was thrown, throw an Error :)
  

  3) devices
       should forget devices:
     Error: the string "requestDevice error: no devices found" was thrown, throw an Error :)
  

  4) services
       "before each" hook for "should get services":
     Error: Timeout of 10000ms exceeded. For async tests and hooks, ensure "done()" is called; if returning a Promise, ensure it resolves. (/Users/ingof/Dev/GitHub/webbluetooth/test/bluetooth.test.js)
      at listOnTimeout (node:internal/timers:573:17)
      at process.processTimers (node:internal/timers:514:7)

  5) services
       "after each" hook for "should get services":
     TypeError: Cannot read properties of undefined (reading 'gatt')
      at Context.<anonymous> (test/bluetooth.test.js:120:22)
      at process.processImmediate (node:internal/timers:478:21)

  6) characteristics
       "before each" hook for "should get characteristics":
     Error: requestDevice error: request in progress
      at BluetoothImpl.requestDevice (dist/bluetooth.js:300:19)
      at Context.<anonymous> (test/bluetooth.test.js:149:34)
      at process.processImmediate (node:internal/timers:478:21)

  7) characteristics
       "after each" hook for "should get characteristics":
     TypeError: Cannot read properties of undefined (reading 'gatt')
      at Context.<anonymous> (test/bluetooth.test.js:156:22)
      at process.processImmediate (node:internal/timers:478:21)

  8) descriptors
       "before each" hook for "should get descriptors":
     Error: requestDevice error: request in progress
      at BluetoothImpl.requestDevice (dist/bluetooth.js:300:19)
      at Context.<anonymous> (test/bluetooth.test.js:248:34)
      at process.processImmediate (node:internal/timers:478:21)

  9) descriptors
       "after each" hook for "should get descriptors":
     TypeError: Cannot read properties of undefined (reading 'gatt')
      at Context.<anonymous> (test/bluetooth.test.js:257:22)
      at process.processImmediate (node:internal/timers:478:21)
```

closed 2024-04-22 by Apollon77

thegecko, 2024-04-22:
Yeah, you need a test device running