thegecko/webbluetooth #242: getDevices() returns an empty list on Windows 11

FLATT-Dev, 2024-08-15:
This code
```
const bluetooth = new Bluetooth({ deviceFound });
const device = await bluetooth.getDevices();
```

returns an empty list and this message in the console

> [INFO] SimpleBLE: D:\a\webbluetooth\webbluetooth\SimpleBLE\simpleble\src\backends\windows\Utils.cpp:33 in initialize_winrt: CoGetApartmentType: cotype=-1, qualifier=0, result=800401F0
> [INFO] SimpleBLE: D:\a\webbluetooth\webbluetooth\SimpleBLE\simpleble\src\backends\windows\Utils.cpp:41 in initialize_winrt: RoInitialize: result=0

The PC runs Windows 11 and has a built in Bluetooth adapter
Any ideas how to fix?
Thank you

thegecko, 2024-08-15:
If possible, can you see if the underlying bluetooth engine works on your system?

https://github.com/OpenBluetoothToolbox/SimpleBLE

FLATT-Dev, 2024-08-15:
Unfortunately I'm not set up to do that at the moment so it will take time.
Looking at the SimpleBLE supported platforms it says "Windows 10+" but does not explicitly list Windows 11 so I'm suspicious. I'll try on Windows 10 and let you know
Thank you!

thegecko, 2024-08-15:
Windows 10 should work, I test on it :)

kilotone, 2025-03-13:
> I am able to reproduce, the same code for discovering at least works from the normal navigator bluetooth implementation https://googlechrome.github.io/samples/web-bluetooth/get-devices.html

Welp, disregard that, I misunderstood the design pattern and was trying to extend the calls into an emitter pattern. Discovered that You need to call bluetooth.requestdevice to populate the devices, then getDevices()

DaryeDev, 2025-06-28:
Bump! Same happening here.

I did try simplepyble, that also uses [SimpleBLE](https://github.com/OpenBluetoothToolbox/SimpleBLE), and worked; but I do need a JS implementation...

thegecko, 2025-06-30:
I need to see some reproducer code for this issue, please

chdh, 2025-11-06:
> I need to see some reproducer code for this issue, please

Here is a simple test case.

[testCase.zip](https://github.com/user-attachments/files/23381421/testCase.zip)

chdh, 2025-11-06:
@jubesab `getDevices()` returns the `devices` array.
The only place where an entry is added to this array is [line 302](https://github.com/thegecko/webbluetooth/blob/dd73f1b7660a8f080dfdac93c2d905747077ea9e/src/bluetooth.ts#L392):

`devices.push(bluetoothDevice);`

This line is only executed when the condition of the IF statement is met:

`if (this.options?.allowAllDevices || this.allowedDevices.has(deviceInfo.id)) {`

`allowAllDevices` is not set in my test case for this issue and the `allowedDevices` set is empty. I don't understand how this could ever work.

thegecko, 2026-02-08:
@FLATT-Dev I've just tried the latest version on Windows 10 and this is working as expected.

If you just want to list devices, you can change your code to:

```javascript
const bluetooth = new Bluetooth({ allowAllDevices: true, scanTime: 5 });
const devices = await bluetooth.getDevices();
console.log(devices);
```
Regarding the messages:

>[INFO] SimpleBLE: D:\a\webbluetooth\webbluetooth\SimpleBLE\simpleble\src\backends\windows\Utils.cpp:33 in initialize_winrt: CoGetApartmentType: cotype=-1, qualifier=0, result=800401F0
[INFO] SimpleBLE: D:\a\webbluetooth\webbluetooth\SimpleBLE\simpleble\src\backends\windows\Utils.cpp:41 in initialize_winrt: RoInitialize: result=0

These have now been removed as unneeded log messages

@chdh Thanks for the test case, it works as expected for me, though. The list of discovered devices wont be returned until the `scantime` is complete, so you need to wait 15 seconds in your case.

>allowAllDevices is not set in my test case for this issue and the allowedDevices set is empty. I don't understand how this could ever work.

This library follows the [webbluetooth specification](https://webbluetoothcg.github.io/web-bluetooth/), so you either need to authorise a device (by calling `requestDevice()` with your filters or `acceptAllDevices`) or create a new Bluetooth class setting the `allowAllDevices` to true. Once either of those are done, `getDevices()` returns authorised (or all) devices it finds

I don't think there is a bug here, so closing the issue. Please re-open as required.

closed 2026-02-08 by thegecko