thegecko/webbluetooth #257: Update license to GPLv3 to account for SimpleBLE new commercial license

jubesab, 2024-10-03:
See https://github.com/OpenBluetoothToolbox/SimpleBLE?tab=readme-ov-file#license for more details.

> Since February 15th 2024, SimpleBLE is now available under the GNU General Public License version 3 (GPLv3), with the option for a commercial license without the GPLv3 restrictions available for a fee.

> The GPLv3 license ensures that end users have the freedom to run, study, share, and modify the software. It requires that any modified versions of SimpleBLE, or any software incorporating it, also be distributed under the GPLv3. Essentially, if your project incorporates SimpleBLE and is distributed, the entire codebase must be open-source under the GPLv3.

jubesab, 2024-10-03:
Indeed, it does seem that 0.6.1 is still under MIT.
But what does this mean for the future of this library though?
You will not be able to get any fix or improvement from SimpleBLE then.

maelp, 2024-10-17:
Would rather keep MIT indeed...