thegecko/webbluetooth #263: Get the MTU

maelp, 2024-10-17:
Is there a way to get the peripheral MTU?

thegecko, 2024-10-19:
Since switching to [SimpleBLE](https://github.com/OpenBluetoothToolbox/SimpleBLE), we are reliant on what that exposes, so that needs cheking.

I'm not sure where we would expose it, I can't see the [Webbluetooth specification](https://webbluetoothcg.github.io/web-bluetooth/) mentions exposing it

maelp, 2024-10-20:
Hmmm... but if you don't expose MTU, how do I know how many bytes can fit in a packet?

thegecko, 2024-10-20:
https://github.com/WebBluetoothCG/web-bluetooth/issues/383

maelp, 2024-10-20:
BTW when using your library in a browser environment, does your library automatically switch to the underlying Webbluetooth backend? Or does the user have to do something like

```
const ble = env.NODE ? import "webbluetooth" : new window.WebBluetooth()
```

or equivalent?

thegecko, 2024-10-20:
This library requires native modules, so would never work in a browser. Ideally you should be creating different builds for web and desktop and depending on your packaging environment you should be able to trivially swap this in when building on desktop. It aims to follow the webbluetooth spec so the API is the same as the native web one.

thegecko, 2025-02-18:
There is a hidden property on the peripheral named `_adData` which isn't in the `webbluetooth` specification. This is used to access advertisement data not present in the spec.

I've exposed the mtu value in that property in: https://github.com/thegecko/webbluetooth/pull/300

thegecko, 2025-03-09:
Released in v3.3.2

closed 2025-03-09 by thegecko