thegecko/webbluetooth #285: startNotifications(): Allow simultaneous listeners

SamuelMereau, 2025-01-03:
Hi all,

A bit of a fascinating usecase... I am currently using this library to interface with external smart glasses which has integrated Bluetooth capability. The hardware device is set up to have two individual BLE devices (one for each side of the glasses), with a UART service bridging both of them. However, due to this, they both share the same characteristic UUID. This is a hinderance as the library only allows notifications to be listened to per characteristic, and forces only one device to be listened to at a time.

Is this an issue relating to the library, or a usecase that is not intended by the spec? 

Many thanks,

thegecko, 2025-01-03:
Interesting use case indeed.

FYI, the spec. is here: https://webbluetoothcg.github.io/web-bluetooth/

I try to follow this (and the chrome implementation) as close as possible.

In theory, you should be able to discover and connect to the individual devices simultaneously. The characteristic instance should then be unique to each device (even though it has the same ID).

If you have trouble doing this with the library, see if it works in chrome. I'd be interested to know of any cases where chrome works and the library doesn't.

SamuelMereau, 2025-01-04:
Hey there, thank you for the quick reply and for linking the Web Bluetooth spec. Hugely appreciate that you keep this implementation in line with the chrome implementation of the web bluetooth spec (matter of fact, was the primary reason why I want to stick with this library... ha!)

I quickly set up a test webpage and copied over my logic for connecting to the devices and establishing connections to the GATT servers and listening to notifications on their respective characteristics. 

For connecting to the devices, I followed this logic:

```
const scanButtonHandler = async (prefix) => {
  const device = await navigator.bluetooth.requestDevice({
      filters: [{ namePrefix: prefix }],
      optionalServices: [SERVICE_UUID],
      acceptAllDevices: false
  }).catch(error => {
      console.error(error);
  });
  
  deviceFound(device);
  console.log("Connected to device: ", device);
};

document.getElementById("scanLeft").addEventListener("click", () => scanButtonHandler(LEFT_PREFIX));
document.getElementById("scanRight").addEventListener("click", () => scanButtonHandler(RIGHT_PREFIX));
```

For every device connected, the deviceFound method is run (this is the equivalent of the `deviceFound` event listener on the `Bluetooth` object in the library, e.g `new Bluetooth({ deviceFound: someDeviceFoundFunc });`

I then establish a connection to the GATT servers like below:

```
if (leftDevice && rightDevice) {
    console.log("Both devices found, connecting to GATT...");

    let connect_tasks = [leftDevice, rightDevice].map(async (device) => {
        try {
            const gattServer = await device.gatt.connect();
            return gattServer;
        } catch (error) {
            throw error;
        }
    });

    (async () => {
        try {
            await Promise.all(connect_tasks);
            console.log("Connected to both devices..!");
        } catch (error) {
            throw error;
        }
    })();
}
```

I can then listen to the characteristics events:

```
const setupDeviceListeners = async (device, side) => {
  device.addEventListener("gattserverdisconnected", () => {
      console.log(`${side} device disconnected`);
  });
  
  try {
      const gattServer = device.gatt;
      const service = await gattServer.getPrimaryService(SERVICE_UUID);
      console.log(`${side} Device Service: `, service);
      const characteristic = await service.getCharacteristic(READ_CHARACTERISTIC_UUID);
      console.log(`${side} Device Characteristic: `, characteristic);
      await characteristic.startNotifications();
      console.log(`${side} Device Notifications started..!`);
      characteristic.addEventListener("characteristicvaluechanged", (event) => {
          console.log(`${side} - Value changed: `, event.target.value);
      });
  } catch (error) {
      console.error(`${side} device setup failed: `, error);
  }
};

if (leftDevice) {
  setupDeviceListeners(leftDevice, "Left");
}

if (rightDevice) {
  setupDeviceListeners(rightDevice, "Right");
}
```

Running this in the Chrome browser yields the exact result I was seeking with the library.

[image]

Doing this same experiment with the library yields the original issue, where only a single device can be listened to.

[image]

FYI I have created a public repo for my usecase utilising the library, so please feel free to observe the code there if you would like something more technical to review: https://github.com/SamuelMereau/evenjs

Thank you again, appreciate your work with this.

thegecko, 2025-01-04:
There's clearly a bug in the library, so thanks for uncovering it.

jvergeldedios, 2025-02-03:
I just ran into this issue and I think I tracked it down to the `SimplebleAdapter` class. The class is capable of discovering and storing several peripherals but when you call `enableNotify` for a given characteristic it adds the handler to the `charEvents` map:
https://github.com/thegecko/webbluetooth/blob/a3a44ac649f38bbe069f1c10a1c26f00fbb59ae5/src/adapters/simpleble-adapter.ts#L331-L333

The problem happens when you call `connect` on a different peripheral it then calls `enumerate` which will clear the `charEvents` map and replace the handlers with the ones for the newly connected device:
https://github.com/thegecko/webbluetooth/blob/a3a44ac649f38bbe069f1c10a1c26f00fbb59ae5/src/adapters/simpleble-adapter.ts#L108-L137

This class probably needs to be reworked so that the discovered services/characteristics/descriptors and characteristic notification handlers get saved on a per-peripheral basis rather than one map that gets cleared on every peripheral connection.

thegecko, 2025-02-03:
Thanks for the investigation. That will really help when I carve out some time to take a look :)

thegecko, 2025-02-17:
I've created a fix over in https://github.com/thegecko/webbluetooth/pull/299

It also includes a new example which connects to multiple devices.

Are you able to test it before I merge?

SamuelMereau, 2025-02-19:
Thanks @thegecko for creating this fix.

I've built this branch locally and plugged it into my project, and I am now able to see events coming from both devices. 👍 

From testing with this version, for an odd reason I keep experiencing the Bluetooth device dropping the connection and sometimes doubling up events... I haven't had the opportunity to fully test my own code with simultaneous events, so due time will tell if its just my code that is borked..!

Until then this looks good to me. 👍

PR #299 (2025-02-25): Fix connecting to multiple devices
Fixes #285