thegecko/webbluetooth #295: Stack corruption error on MacOS

jvergeldedios, 2025-02-07:
I can't seem to find a pattern, but I intermittently get a stack corruption error that crashes my node process with a SIGABRT. I went into the console and found the following backtrace:

```Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x18a46aa60 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x18a4a2c20 pthread_kill + 288
2   libsystem_c.dylib             	       0x18a3afac4 __abort + 136
3   libsystem_c.dylib             	       0x18a3a125c __stack_chk_fail + 96
4   simpleble.node                	       0x11c09280c Peripheral::GetManufacturerData(Napi::CallbackInfo const&) + 632
5   simpleble.node                	       0x11c094710 napi_value__* Napi::details::TemplatedInstanceCallback<Peripheral, &Peripheral::GetManufacturerData(Napi::CallbackInfo const&)>(napi_env__*, napi_callback_info__*)::'lambda'()::operator()() const + 100
6   simpleble.node                	       0x11c094650 napi_value__* Napi::details::TemplatedInstanceCallback<Peripheral, &Peripheral::GetManufacturerData(Napi::CallbackInfo const&)>(napi_env__*, napi_callback_info__*) + 40
7   node                          	       0x1047c72a8 v8impl::(anonymous namespace)::FunctionCallbackWrapper::Invoke(v8::FunctionCallbackInfo<v8::Value> const&) + 72
8   node                          	       0x1053f6f98 Builtins_CallApiCallbackGeneric + 184
9   node                          	       0x105434d88 Builtins_LoadIC + 3432
10  ???                           	       0x10c149214 ???
11  node                          	       0x1053f4ef0 Builtins_InterpreterEntryTrampoline + 272
12  node                          	       0x1053f2c0c Builtins_JSEntryTrampoline + 172
13  node                          	       0x1053f28f4 Builtins_JSEntry + 148
14  node                          	       0x104b8a728 v8::internal::(anonymous namespace)::Invoke(v8::internal::Isolate*, v8::internal::(anonymous namespace)::InvokeParams const&) + 1676
15  node                          	       0x104b8a06c v8::internal::Execution::Call(v8::internal::Isolate*, v8::internal::Handle<v8::internal::Object>, v8::internal::Handle<v8::internal::Object>, int, v8::internal::Handle<v8::internal::Object>*) + 120
16  node                          	       0x104a5ae8c v8::Function::Call(v8::Local<v8::Context>, v8::Local<v8::Value>, int, v8::Local<v8::Value>*) + 532
17  node                          	       0x1047c1788 napi_call_function + 224
18  simpleble.node                	       0x11c08ac44 Napi::Function::Call(std::initializer_list<napi_value__*> const&) const + 68
19  simpleble.node                	       0x11c08b1c0 std::__1::__function::__func<napi_status Napi::ThreadSafeFunction::NonBlockingCall<void, Adapter::onScanFound(void*, void*, void*)::$_0>(void*, Adapter::onScanFound(void*, void*, void*)::$_0) const::'lambda'(Napi::Env, Napi::Function), std::__1::allocator<napi_status Napi::ThreadSafeFunction::NonBlockingCall<void, Adapter::onScanFound(void*, void*, void*)::$_0>(void*, Adapter::onScanFound(void*, void*, void*)::$_0) const::'lambda'(Napi::Env, Napi::Function)>, void (Napi::Env, Napi::Function)>::operator()(Napi::Env&&, Napi::Function&&) + 128
20  simpleble.node                	       0x11c08ab3c void Napi::details::WrapVoidCallback<Napi::ThreadSafeFunction::CallJS(napi_env__*, napi_value__*, void*, void*)::'lambda'()>(napi_env__*, Napi::ThreadSafeFunction::CallJS(napi_env__*, napi_value__*, void*, void*)::'lambda'()) + 76
21  simpleble.node                	       0x11c08aae4 Napi::ThreadSafeFunction::CallJS(napi_env__*, napi_value__*, void*, void*) + 56
22  node                          	       0x1047ddaf0 v8impl::(anonymous namespace)::ThreadSafeFunction::AsyncCb(uv_async_s*) + 464
23  node                          	       0x1053d2900 uv__async_io + 276
24  node                          	       0x1053e62f4 uv__io_poll + 1024
25  node                          	       0x1053d2e98 uv_run + 412
26  node                          	       0x1047364e4 node::SpinEventLoopInternal(node::Environment*) + 256
27  node                          	       0x104860628 node::NodeMainInstance::Run(node::ExitCode*, node::Environment*) + 184
28  node                          	       0x1048603c4 node::NodeMainInstance::Run() + 124
29  node                          	       0x1047d6f88 node::Start(int, char**) + 652
30  dyld                          	       0x18a11a0e0 start + 2360
```

It looks like it's happening in the Peripheral::GetManufacturerData method. I tried to add logging to see if anything would come out at the beginning of the method before the stack corruption occurs, but nothing was logging. I also tried updating SimpleBLE since the backtrace mentions `simpleble.node` to 0.9.0 and 0.8.1 but the behavior remains.

Relevant version info:
webbluetooth: 3.2.1
MacOS: 14.5

jvergeldedios, 2025-02-08:
Never mind on the multi-device theory. I've seen it happen with a single device multiple times now.

thegecko, 2025-02-11:
I'm keen to know if this is an issue with SimpleBLE or this wrapper. Do you have any data on whether this issue occurs if using SimpleBLE directly?

jvergeldedios, 2025-02-13:
I haven't tried that. I'll see if I can find some time to create a minimal reproduction of both and link it here.

maelp, 2026-02-26:
I also have errors on osX, 

```
libc++abi: terminating due to uncaught exception of type Napi::Error:
 ELIFECYCLE  Command failed with exit code 134.
```

but not much more, how can I debug to understand?