thegecko/webbluetooth #329: TypeScript type error for BluetoothRemoteGATTCharacteristicImpl.uuid

chdh, 2025-05-14:
When compiling with TypeScript 5.8, I get the following error message:

```
node_modules/webbluetooth/dist/characteristic.d.ts:23:14 - error TS2416:
Property 'uuid' in type 'BluetoothRemoteGATTCharacteristicImpl' is not assignable to the same property in base type 'BluetoothRemoteGATTCharacteristic'.
Type 'string | undefined' is not assignable to type 'string'.
 Type 'undefined' is not assignable to type 'string'.
  readonly uuid: string | undefined;
Found 1 error in node_modules/webbluetooth/dist/characteristic.d.ts:23
```

The source of the problem is that the `uuid` field in `BluetoothRemoteGATTCharacteristic` in `@types\web-bluetooth\index.d.ts` is declared as `string`, but in the derived class `BluetoothRemoteGATTCharacteristicImpl` in `webbluetooth\dist\characteristic.d.ts` it's declared as `string | undefined`.

thegecko, 2025-05-14:
Hmm, checked this locally and TypeScript `5.8.3` (the version in the lock file) works fine on MacOS. Have you modified any of the source or dependencies?

chdh, 2025-05-15:
I also use TypeScript 5.8.3 and haven't modified any of the sources.
To prevent the compilation error, I now added the following option in tsconfig.json:

`"skipLibCheck": true`

[testCase.zip](https://github.com/user-attachments/files/20217357/testCase.zip)

PR #334 (2025-05-19): Fix characteristic uuid type
Fixes #329
closed 2025-05-19 by thegecko