thegecko/webbluetooth #330: Annoying log output on the console when running under Windows.

chdh, 2025-05-15:
When Webbluetooth runs under Windows and initializes the SimpleBLE library, the following two lines appear on the console:

```
[INFO] SimpleBLE: D:\a\webbluetooth\webbluetooth\SimpleBLE\simpleble\src\backends\windows\Utils.cpp:33 in initialize_winrt: CoGetApartmentType: cotype=-1, qualifier=0, result=800401F0
[INFO] SimpleBLE: D:\a\webbluetooth\webbluetooth\SimpleBLE\simpleble\src\backends\windows\Utils.cpp:41 in initialize_winrt: RoInitialize: result=0
```

I couldn't find a way to prevent that (without re-compiling the SimpleBLE library). Since I want to build a command line tool, I have to get rid of these lines.

The logging output stems from the following two statements in `SimpleBLE\simpleble\src\backends\windows\Utils.cpp`:

```
    SIMPLEBLE_LOG_INFO(fmt::format("CoGetApartmentType: cotype={}, qualifier={}, result={:X}", cotype, qualifier,
                                   (uint32_t)get_apartment_result));
    ...
    SIMPLEBLE_LOG_INFO(fmt::format("RoInitialize: result={:X}", (uint32_t)result));
```

In the [newest version of Utils.cpp](https://github.com/simpleble/simpleble/blob/main/simpleble/src/backends/windows/Utils.cpp), these log output statements have been corrected.

thegecko, 2025-05-19:
https://github.com/thegecko/webbluetooth/issues/257
https://github.com/thegecko/webbluetooth/tree/simpleble-upgrade

chdh, 2025-05-21:
Could you remove the two `SIMPLEBLE_LOG_INFO` lines in [your fork of SimpleBLE](https://github.com/thegecko/simpleble)?