thegecko/webbluetooth #331: getDevices() does not resolve device names when called with allowAllDevices: true

chdh, 2025-05-15:
When I call `bluetooth.getDevices()` with `options.allowAllDevices: true`, the device names are not retrieved:

```
Found device: Unknown or Unsupported Device (7f:6c:f3:b7:88:e9)
Found device: Unknown or Unsupported Device (7a:25:30:8c:e2:71)
Found device: Unknown or Unsupported Device (56:eb:35:12:42:40)
Found device: Unknown or Unsupported Device (cb:51:ab:53:8d:fa)
Found device: Unknown or Unsupported Device (d2:00:05:0c:f2:1c)
```

When I set`allowAllDevices` to `false`, `getDevices()` does not return any devices because the `allowedDevices` set is empty in [bluetooth.ts](https://github.com/thegecko/webbluetooth/blob/main/src/bluetooth.ts).

But when I first call `requestDevice({acceptAllDevices: true})` before calling `bluetooth.getDevices()` with `options.allowAllDevices: true`, the name of the device that was previously returned by `requestDevice()` is retrieved in the list returned by `bluetooth.getDevices()`.

[testCase.zip](https://github.com/user-attachments/files/20217782/testCase.zip)

jubesab, 2025-11-05:
Hi @chdh, does this issue persist? Using your test case, I'm not having any. Supported devices like my Jabra headset and other BLE devices are showing up. Tested on Win 11 Pro 24H2.

chdh, 2025-11-06:
@jubesab Yes, it's still the same with webbluetooth version 3.3.2.

jubesab, 2025-11-06:
@chdh I'm using your test case so I'm running 3.3.2 also. What OS and version are you using? So I can try to replicate the issue.

chdh, 2025-11-06:
@jubesab I'm using Windows 11 (24H2, 10.0.26100.6899).
Intel Bluetooth driver version 23.170.0.3.